tesseract-robotics/tesseract_planning #437: IKFast and QPOases Incompatibility

Levi-Armstrong, 2024-02-07:
It looks like for some reason when qpoases is installed from source from trajopt_ext when dynamically loading ikfast kinematics plugins it uses the `build/qpoases-src/src/BLASReplacement.cpp` causing the planning server to crash with memory alignment issues. 

``` bash
[Switching to Thread 0x7fffbf7fe700 (LWP 95468)]
__memset_avx2_erms () at ../sysdeps/x86_64/multiarch/memset-vec-unaligned-erms.S:151
151 ../sysdeps/x86_64/multiarch/memset-vec-unaligned-erms.S: No such file or directory.
(gdb) bt
#0  __memset_avx2_erms () at ../sysdeps/x86_64/multiarch/memset-vec-unaligned-erms.S:151
#1  0x00007fffe9906ac1 in dgemm_(char const*, char const*, la_uint_t const*, la_uint_t const*, la_uint_t const*, double const*, double const*, la_uint_t const*, double const*, la_uint_t const*, double const*, double*, la_uint_t const*)
    (TRANSA=0x7fff68f4ad19 "N", TRANSB=<optimized out>, M=0x7fffbf7f0e48, N=<optimized out>, K=0x7fffbf7f0968, ALPHA=0x7fff68f4da28, A=<optimized out>, LDA=0x7fffbf7f0e48, B=0x7fffbf7f5730, LDB=0x7fffbf7f0e48, BETA=0x7fff68f4da20, C=0x7fffbf7f92b0, LDC=0x7fffbf7f0e48)
    at <workspace>/build/qpoases-src/src/BLASReplacement.cpp:49
#2  0x00007fff68c582b5 in dtrevc3_ () at /lib/x86_64-linux-gnu/liblapack.so.3
#3  0x00007fff68b1bf12 in dgeev_ () at /lib/x86_64-linux-gnu/liblapack.so.3
#4  0x00007fff7079c67d in IKSolver::solvedialyticpoly8qep(double const*, double*, int&)
    (matcoeffs=0x7fffbf7fa350, rawroots=0x7fffbf7fa1d0, numroots=@0x7fffbf7f9b00: 0)
    at ...robot_ikfast.hpp:7282
#5  0x0000000000000000 in  ()
```

rjoomen, 2024-02-07:
Maybe related: https://github.com/ros-industrial-consortium/scan_n_plan_workshop/pull/22#issuecomment-1603306370

Levi-Armstrong, 2024-02-07:
It has been confirmed that if you delete the qpoases directory from trajopt_ext and clean rebuild the issue goes away.

rjoomen, 2024-02-19:
For reference: qpOASES is properly linked as private [here](https://github.com/tesseract-robotics/trajopt/blob/master/trajopt_sco/CMakeLists.txt#L105).