tesseract-robotics/tesseract_planning #453: The timing of enforcePositionLimits function call in OMPLDefaultPlanProfile::applyStartStates

MinRen86, 2024-04-29:
In function [`OMPLDefaultPlanProfile::applyStartStates`](https://github.com/tesseract-robotics/tesseract_planning/blob/28752ea64738aeea1034039e9eb747bfbb5e1fe6/tesseract_motion_planners/ompl/src/profile/ompl_default_plan_profile.cpp#L593C6-L593C46).

Why is [`tesseract_common::enforcePositionLimits<double>(solution, limits.joint_limits);`](https://github.com/tesseract-robotics/tesseract_planning/blob/28752ea64738aeea1034039e9eb747bfbb5e1fe6/tesseract_motion_planners/ompl/src/profile/ompl_default_plan_profile.cpp#L609C7-L609C86) called after the [`tesseract_common::satisfiesPositionLimits<double>(solution, limits.joint_limits)`](https://github.com/tesseract-robotics/tesseract_planning/blob/28752ea64738aeea1034039e9eb747bfbb5e1fe6/tesseract_motion_planners/ompl/src/profile/ompl_default_plan_profile.cpp#L607-614) is satisfied, rather than when it is not satisfied?

The entire paragraph is below:
```
void OMPLDefaultPlanProfile::applyStartStates(OMPLProblem& prob,
                                              const Eigen::VectorXd& joint_waypoint,
                                              const MoveInstructionPoly& /*parent_instruction*/,
                                              const tesseract_common::ManipulatorInfo& /*manip_info*/,
                                              const std::vector<std::string>& /*active_links*/,
                                              int /*index*/) const
{
    // ...
    if (tesseract_common::satisfiesPositionLimits<double>(solution, limits.joint_limits))
    {
      tesseract_common::enforcePositionLimits<double>(solution, limits.joint_limits);
    }
    else
    {
      CONSOLE_BRIDGE_logDebug("In OMPLDefaultPlanProfile: Start state is outside limits");
    }
    // ...
}
```

What I imagined:
```
void OMPLDefaultPlanProfile::applyStartStates(OMPLProblem& prob,
                                              const Eigen::VectorXd& joint_waypoint,
                                              const MoveInstructionPoly& /*parent_instruction*/,
                                              const tesseract_common::ManipulatorInfo& /*manip_info*/,
                                              const std::vector<std::string>& /*active_links*/,
                                              int /*index*/) const
{
    // ...
    if (!tesseract_common::satisfiesPositionLimits<double>(solution, limits.joint_limits))
    {
      tesseract_common::enforcePositionLimits<double>(solution, limits.joint_limits);
      CONSOLE_BRIDGE_logDebug("In OMPLDefaultPlanProfile: Start state is outside limits");
    }
    // ...
}
```

Levi-Armstrong, 2024-04-29:
The current state is intended because it solves numerical issues. The `satisfiesPositionLimits` check is tolerance based so it could be slightly outside the limits and the enforce ensure if slightly outside the limit if forces it to the limit to avoid this numerical issue.

MinRen86, 2024-04-30:
Got it !! Thanks a lot !

closed 2024-04-30 by MinRen86