tesseract-robotics/tesseract_planning #454: Joint names for waypoints need to be ordered

dave992, 2024-04-30:
I ran into some assertion errors when creating a plan using waypoints created from information from the SRDF. To be more specific, using the `tesseract_environment::Environment` I can get a group joint state from the SRDF:
```c++
std::string group_name("manipulator");
std::string state_name("home");

auto kinematic_info = env->getKinematicsInformation();
std::unordered_map<std::string, double> group_joint_state = kinematics_info.group_states[group_name][state_name];
```

This information is used to create a `tesseract_planning::JointWaypoint`.  I would expect it does not matter in what order I define the joint names and their values, as long as I define them all.

Using this approach I sometimes run into assertion errors when creating a plan. More specifically here:  [tesseract_motion_planners/trajopt/src/trajopt_motion_planner.cpp#L280](https://github.com/tesseract-robotics/tesseract_planning/blob/4db71a632d3c9264d7a60f6ce1c2317d7cb12295/tesseract_motion_planners/trajopt/src/trajopt_motion_planner.cpp#L280):
```c++
assert(checkJointPositionFormat(joint_names, move_instruction.getWaypoint()));
```

Inspecting the `checkJointPositionFormat` function, I see that the `joint_names` are compared to the names defined in the waypoint:  [tesseract_command_language/src/utils.cpp#L280-L296](https://github.com/tesseract-robotics/tesseract_planning/blob/28752ea64738aeea1034039e9eb747bfbb5e1fe6/tesseract_command_language/src/utils.cpp#L280-L296): 
```c++
bool checkJointPositionFormat(const std::vector<std::string>& joint_names, const WaypointPoly& waypoint)
{
  if (waypoint.isJointWaypoint())
    return (joint_names == waypoint.as<JointWaypointPoly>().getNames());

  if (waypoint.isStateWaypoint())
    return (joint_names == waypoint.as<StateWaypointPoly>().getNames());

  if (waypoint.isCartesianWaypoint())
  {
    const auto& cwp = waypoint.as<CartesianWaypointPoly>();
    if (cwp.hasSeed())
      return (joint_names == waypoint.as<CartesianWaypointPoly>().getSeed().joint_names);
  }

  throw std::runtime_error("Unsupported waypoint type.");
}
```

Here the `joint_names` parameter seems to be retrieved from the kinematic info. Since it is using the `==` operator from the `std::vector`, it check if the vectors are equal, not if they contain the same information. 

### Expected behavior
The order of the `joint_names` does not matter, at least for the `JointWaypoint`, as long as they correspond to the correct values. 

### Additional questions
- How is the **_correct_** order of the `joint_names` defined? Is this just the order as they appear in the URDF/SRDF?
- How do you recommend to retrieve the correct `joint_names` order? I currently use the following:
  ```c++
    std::vector<std::string> joint_names = env->getKinematicsInformation().joint_groups[group_name_];
  ```

Levi-Armstrong, 2024-04-30:
This is intended behavior for performance reason. Forward and Inverse kinematics is called thousands of time during a motion planning and support unordered data at the low level would be a significant performance hit. Instead there is a task which you can added to your pipeline which will format the program to correctly work with motion planning. This way you just pay the penalty once.

dave992, 2024-04-30:
Thanks for the answer. Is this the `CheckInputTask` you are referring to?

Levi-Armstrong, 2024-04-30:
Well I was wrong about the task existing but the following utility function exists in tesseract_motion_planners utils.cpp to format the input. I will work on creating a task which leverages this.

```bool formatProgram(CompositeInstruction& composite_instructions, const tesseract_environment::Environment& env)```

dave992, 2024-04-30:
Thanks, I will have a look at the function!

aizmailovs, 2024-11-12:
I am also running into this assertion.
For context, I am trying to realize the [scan-n-plan workshop](https://github.com/ros-industrial-consortium/scan_n_plan_workshop) with UR robot arm. However, the joint order differs between joint group and start waypoint.

> Well I was wrong about the task existing but the following utility function exists in tesseract_motion_planners utils.cpp to format the input. I will work on creating a task which leverages this. [...]

How to apply this utility function?

dave992, 2024-11-12:
> I am also running into this assertion. For context, I am trying to realize the [scan-n-plan workshop](https://github.com/ros-industrial-consortium/scan_n_plan_workshop) with UR robot arm. However, [...]

Make sure you add an import for the [`formatProgram`](https://github.com/tesseract-robotics/tesseract_planning/blob/e5f06ebfdfa0eb6000ffc6d88f6147ecd2cf51a4/tesseract_motion_planners/core/src/utils.cpp#L281) function. 

```c++
// Create a 'program' to plan for
tesseract_planning::CompositeInstruction program( ... );
      
// Add move instructions (with a joint waypoint) to your program
program.appendMoveInstruction( ... );

tesseract_planning::formatProgram(program, env); # I assume you have a tesseract_environement::Environment object already

// your program is now formatted and can be used for planning
```

Levi-Armstrong, 2024-11-13:
The latest commit on this repository adds a task to format the program correctly. Also all of the pipelines in the yaml file have been updated to include this task.

Levi-Armstrong, 2024-11-17:
Addressed by this [PR](https://github.com/tesseract-robotics/tesseract_planning/pull/530)

closed 2024-11-17 by Levi-Armstrong

aizmailovs, 2024-11-25:
Thank you @dave992 and @Levi-Armstrong for your input!

> Make sure you add an import for the [formatProgram](https://github.com/tesseract-robotics/tesseract_planning/blob/e5f06ebfdfa0eb6000ffc6d88f6147ecd2cf51a4/tesseract_motion_planners/core/src/utils.cpp#L281) function.

@dave992, I have not tried it yet, but it looks like a valid solution, I appreciate it!

> The latest commit on this repository adds a task to format the program correctly. Also all of the pipelines in the yaml file have been updated to include this task.

> Addressed by this https://github.com/tesseract-robotics/tesseract_planning/pull/530

@Levi-Armstrong, this is exactly what I was hoping for, thank you! Unfortunately, I [could not yet integrate it into scan-n-plan workshop](https://github.com/ros-industrial-consortium/scan_n_plan_workshop/discussions/147), likely due to dependency on earlier tesseract versions. I will either attempt to manually integrate this task into the earlier version of tesseract or wait for the update of scan-n-plan workshop.

aizmailovs, 2024-12-09:
I have managed to integrate the task into scan-n-plan workshop and it is working! Task pipeline yaml file was very useful, thanks!
[image]