tesseract-robotics/tesseract_planning #480: Hi, How can i solve this problem ContactManagersPluginFactory, tried to get discrete contact manager '' that does not exist!

Jm20010201, 2024-07-24:
[image]

I want to use your planning library to plan the motion of a mobile manipulator. Here is my code, and the above picture shows the problem I encountered. Can you help me see what the reason is? Thank you！！！

`#include <iostream>

#include <tesseract_common/macros.h>
TESSERACT_COMMON_IGNORE_WARNINGS_PUSH
#include <console_bridge/console.h>
TESSERACT_COMMON_IGNORE_WARNINGS_POP

#include <tesseract_examples/whole_body_plan.h>

#include <tesseract_common/timer.h>

#include <tesseract_kinematics/core/utils.h>

#include <tesseract_state_solver/state_solver.h>

#include <tesseract_environment/environment.h>
#include <tesseract_environment/utils.h>
#include <tesseract_environment/commands/add_link_command.h>
#include <tesseract_collision/core/types.h>

#include <tesseract_command_language/state_waypoint.h>
#include <tesseract_command_language/cartesian_waypoint.h>
#include <tesseract_command_language/joint_waypoint.h>
#include <tesseract_command_language/move_instruction.h>
#include <tesseract_command_language/utils.h>
#include <tesseract_command_language/profile_dictionary.h>

#include <tesseract_visualization/markers/toolpath_marker.h>
#include <tesseract_visualization/visualization.h>
#include <tesseract_visualization/visualization_loader.h>
#include <tesseract_support/tesseract_support_resource_locator.h>

#include <tesseract_motion_planners/core/types.h>
#include <tesseract_motion_planners/core/utils.h>
#include <tesseract_motion_planners/simple/interpolation.h>
#include <tesseract_motion_planners/ompl/ompl_planner_configurator.h>

#include <tesseract_motion_planners/trajopt/trajopt_motion_planner.h>
#include <tesseract_motion_planners/trajopt/profile/trajopt_default_plan_profile.h>
#include <tesseract_motion_planners/trajopt/profile/trajopt_default_composite_profile.h>

#include <tesseract_motion_planners/ompl/profile/ompl_default_plan_profile.h>
#include <tesseract_motion_planners/ompl/ompl_motion_planner.h>

#include <tesseract_support/tesseract_support_resource_locator.h>

using namespace tesseract_planning;
using tesseract_common::ManipulatorInfo;

static const std::string TRAJOPT_IFOPT_DEFAULT_NAMESPACE = "TrajOptIfoptMotionPlannerTask";
static const std::string TRAJOPT_DEFAULT_NAMESPACE = "TrajOptMotionPlannerTask";
const static std::string OMPL_DEFAULT_NAMESPACE = "OMPLMotionPlannerTask";

namespace tesseract_examples
{

WholeBodyPlan::WholeBodyPlan(std::shared_ptr<tesseract_environment::Environment> env,
                                             std::shared_ptr<tesseract_visualization::Visualization> plotter,
                                             bool ifopt,
                                             bool debug)
  : Example(std::move(env), std::move(plotter)), ifopt_(ifopt), debug_(debug)
{
}

bool WholeBodyPlan::run()
{
  tesseract_visualization::VisualizationLoader loader;
  auto plotter_ = loader.get();
  if (plotter_ != nullptr)
    plotter_->waitForConnection();

  // Set the robot initial state
  std::vector<std::string> joint_names;
  joint_names.emplace_back("base_link_base_theta");
  joint_names.emplace_back("base_theta_base_y");
  joint_names.emplace_back("base_y_base_x");
  joint_names.emplace_back("elbow_joint");
  joint_names.emplace_back("shoulder_lift_joint");
  joint_names.emplace_back("shoulder_pan_joint");
  joint_names.emplace_back("wrist_1_joint");
  joint_names.emplace_back("wrist_2_joint");
  joint_names.emplace_back("wrist_3_joint");

  Eigen::VectorXd joint_pos(9);
  joint_pos(0) = 0;
  joint_pos(1) = 0;
  joint_pos(2) = 0.0;
  joint_pos(3) = -0.0425;
  joint_pos(4) = 0.6096;
  joint_pos(5) = 0.25037;
  joint_pos(6) = 1.061965;
  joint_pos(7) = 1.321897;
  joint_pos(8) = -0.000931;

  env_->setState(joint_names, joint_pos);

  if (debug_)
    console_bridge::setLogLevel(console_bridge::LogLevel::CONSOLE_BRIDGE_LOG_DEBUG);
  else
    console_bridge::setLogLevel(console_bridge::LogLevel::CONSOLE_BRIDGE_LOG_INFO);

  tesseract_common::ManipulatorInfo manip;
  manip.tcp_frame = "wrist_3_joint";
  manip.working_frame = "world";
  manip.manipulator = "manipulator";
  //manip.manipulator_ik_solver = "OPWInvKin";

  auto state_solver = env_->getStateSolver();
  //auto kin_group = env_->getKinematicGroup(manip.manipulator, manip.manipulator_ik_solver);
  auto cur_state = env_->getState();

  
  StateWaypointPoly wp0{ StateWaypoint(joint_names, joint_pos) };
  joint_pos(0) = 0;
  joint_pos(2) = 2;
  StateWaypointPoly wpend{ StateWaypoint(joint_names, joint_pos) };

  MoveInstruction start_instruction(wp0, MoveInstructionType::FREESPACE, "DEFAULT");
  MoveInstruction plan_f1(wpend, MoveInstructionType::FREESPACE, "DEFAULT");

      // Create program
  CompositeInstruction program;
  program.setManipulatorInfo(manip); 
  program.appendMoveInstruction(start_instruction);
  program.appendMoveInstruction(plan_f1);

  // Print Diagnostics
  program.print("Program: ");

  CONSOLE_BRIDGE_logInform("whole body plan example");

   // Create Profiles
    auto ompl_plan_profile = std::make_shared<OMPLDefaultPlanProfile>();
    auto ompl_planner_config = std::make_shared<RRTConnectConfigurator>();
    ompl_plan_profile->planning_time = 10;
    ompl_plan_profile->planners = { ompl_planner_config, ompl_planner_config };

    auto trajopt_plan_profile = std::make_shared<TrajOptDefaultPlanProfile>();
    

    auto trajopt_composite_profile = std::make_shared<TrajOptDefaultCompositeProfile>();

    // Create a interpolated program
    CompositeInstruction interpolated_program = generateInterpolatedProgram(program, cur_state, env_);

    // Profile Dictionary
    auto profiles = std::make_shared<ProfileDictionary>();
    profiles->addProfile<OMPLPlanProfile>(OMPL_DEFAULT_NAMESPACE, "DEFAULT", ompl_plan_profile);
    profiles->addProfile<TrajOptPlanProfile>(TRAJOPT_DEFAULT_NAMESPACE, "DEFAULT", trajopt_plan_profile);
    profiles->addProfile<TrajOptCompositeProfile>(TRAJOPT_DEFAULT_NAMESPACE, "DEFAULT", trajopt_composite_profile);

    // Create Planning Request
    PlannerRequest request;
    request.instructions = interpolated_program;
    request.env = env_;
    request.env_state = cur_state;
    request.profiles = profiles;
    

    tesseract_common::Timer stopwatch;
    stopwatch.start();
    // Solve OMPL Plan
    OMPLMotionPlanner ompl_planner(OMPL_DEFAULT_NAMESPACE);
    PlannerResponse ompl_response = ompl_planner.solve(request);
    assert(ompl_response);
    
    stopwatch.stop();
    CONSOLE_BRIDGE_logInform("ompl Planning took %f seconds.", stopwatch.elapsedSeconds());

    // Plot OMPL Trajectory
    if (plotter_)
    {
      plotter_->waitForInput();
      plotter_->plotTrajectory(toJointTrajectory(ompl_response.results), *state_solver);
    }

    // Update Seed
    request.instructions = ompl_response.results;
    
    stopwatch.start();
    // Solve TrajOpt Plan
    TrajOptMotionPlanner trajopt_planner(TRAJOPT_DEFAULT_NAMESPACE);
    PlannerResponse trajopt_response = trajopt_planner.solve(request);
    assert(trajopt_response);
    stopwatch.stop();
    CONSOLE_BRIDGE_logInform("trajopt Planning took %f seconds.", stopwatch.elapsedSeconds());
    if (plotter_)
    {
      plotter_->waitForInput();
      plotter_->plotTrajectory(toJointTrajectory(trajopt_response.results), *state_solver);
    }

    CONSOLE_BRIDGE_logInform("Final trajectory is collision free");
    return true;
}

}  // namespace tesseract_examples
`

Levi-Armstrong, 2024-07-24:
What version of the software are you using?

Jm20010201, 2024-07-24:
> What version of the software are you using?

master

Jm20010201, 2024-07-24:
Hi，I want to use SLAM to build a point cloud map, and then add the point cloud to env like in the Basic cartesian example to plan the full body motion of the mobile manipulator. Do you think this is feasible?  Now I have encountered the above problem, do you know how to solve it？

Levi-Armstrong, 2024-07-24:
Did you build tesseract_collision bullet and fcl?

Jm20010201, 2024-07-24:
> Did you build tesseract_collision bullet and fcl?

Yes, I have built the Tesseract, Collision, Bullet, and FCL libraries. I can execute the file in examples. I guess my code is missing a certain setting. Could you please help me take a look?

closed 2024-07-25 by Levi-Armstrong