tesseract-robotics/tesseract_planning #488: Hello, there are nine joints in my planning group, including the six joints of the robotic arm and the base. How should I modify the freespace example?

Jm20010201, 2024-07-27:
I want to use hybrid A star to plan the path of the chassis. I found that the previous version of Tesseract planning had 3motionplanning, which can realize simultaneous planning of base and arm. How should I modify it in the current version?

Levi-Armstrong, 2024-07-27:
If you want to create and example of what you want and create a pull request, I would be happy to work with you on getting it working. Something like this is challenging to provide recommendation without an example to work with.

Jm20010201, 2024-07-29:
> If you want to create and example of what you want and create a pull request, I would be happy to work with you on getting it working. Something like this is challenging to provide recommendation without an example to work with.

Thank you very much!!! I have already created a pull request.