tesseract-robotics/tesseract_planning #499: Failed edges and vertices with Descartes when using fixed target pose sampler

srsidd, 2024-08-06:
When using the Descartes planner, I'm noticing some strange behavior. 

Creating a `DescartesDefaultPlanProfile` with the target pose sampler set to any one of `sampleToolXAxis`, `sampleToolYAxis` or `sampleToolZAxis` results in the graph being built correctly. The planner goes through and I get a motion plan. 

However, in my appliaction I cannot have a rotation offset in any axes, so I tried not setting the target_pose_sampler (which defaults to the `sampleFixed` function). When I do this, Descartes throws an error with failed edges and vertices. I tried setting `allow_collision` to see if that might be causing an issue but that had no impact. 

Any idea what maybe going on here / any pointers on how to debug this?

marip8, 2024-08-06:
It seems like your nominal waypoints do not have a valid IK solution. They might be reachable if you change the orientation of the nominal waypoint (e.g., via the sampling you originally performed). If `allow_collision=true` and you still don't get any vertices, then it seems like no IK solutions were found. 

You could confirm by attaching a debugger to the application and looking at the output of [this line](https://github.com/tesseract-robotics/tesseract_planning/blob/e41b8374cec23e18b8221d6135304adc2d54ee78/tesseract_motion_planners/descartes/include/tesseract_motion_planners/descartes/impl/descartes_robot_sampler.hpp#L93). If valid IK solutions exist, this vector should not be empty

srsidd, 2024-08-06:
I forgot to mention, that the planner fails only for if descartes is enabled. If I skipped Descartes as a global planner, and use TrajOpt directly, the planner succeeds and is able to get the correct plan (without the resampled offset)

marip8, 2024-08-06:
Does your `trajopt` profile have custom weights for the Cartesian waypoints where one axis is set to zero (e.g., [5.0, 5.0, 5.0, 5.0, 5.0, 0.0])? If so, it's possible that `trajopt` is changing the position/orientation of your Cartesian waypoints to make them kinematically feasible.

srsidd, 2024-08-07:
No, the cartesian coefficients are all set to 1,  `profile->cartesian_coeff = Eigen::VectorXd::Constant(6, 1, 1.0);` and no joint coefficients have been specified.

marip8, 2024-08-07:
That is a bit strange then. What version/commite of `descartes` and `tesseract_planning` are you using? Can you upload some of the planning artifacts (e.g., pipeline graphviz files, etc.) and terminal log to this issue? You probably will need to build and run in debug to get the full output from the planners

srsidd, 2024-08-28:
I'm going to close this issue, I am unable to recreate this behavior. It might have been user error. I will re-open if I notice this behavior again.

closed 2024-08-28 by srsidd