tesseract-robotics/tesseract_planning #509: windows build fail

xbkaishui, 2024-08-24:
Hello:

I am building planning master branch on win11, follow the build instruction in github action windows.yml,  failed at build tesseract_task_composer 

details as follows: 
command 

> >colcon build --packages-select tesseract_task_composer --packages-up-to tesseract_python --merge-install   --cmake-force-configure   --event-handlers console_cohesion+   --packages-ignore tesseract_examples trajopt_ifopt trajopt_sqp gtest   --cmake-args -GNinja -DBUILD_IPOPT=OFF -DBUILD_SNOPT=OFF -DTESSERACT_BUILD_TRAJOPT_IFOPT=OFF -DVCPKG_APPLOCAL_DEPS=OFF -DTESSERACT_BUILD_DESCARTES=OFF -DCMAKE_BUILD_TYPE=Release   -DPYTHON_EXECUTABLE="C:\Users\xxkai\miniconda3\envs\rac\python.exe"   -DTESSERACT_PYTHON_BUILD_WHEEL=ON   -DTESSERACT_ENABLE_EXAMPLES=OFF -DTESSERACT_PLUGIN_FACTORY_CALLBACKS=ON   -DVCPKG_APPLOCAL_DEPS=OFF   -DTESSERACT_BUILD_TRAJOPT_IFOPT=OFF

error log 
 

> 

-- Generating done (0.3s)
-- Build files have been written to: C:/dev/tract/build/tesseract_task_composer
[1/12] Building CXX object planning\CMakeFiles\tesseract_task_composer_planning_factories.dir\src\factories\planning_task_composer_ruckig_plugin_factory.cpp.obj
FAILED: planning/CMakeFiles/tesseract_task_composer_planning_factories.dir/src/factories/planning_task_composer_ruckig_plugin_factory.cpp.obj
C:\PROGRA~2\MICROS~2\2019\PROFES~1\VC\Tools\MSVC\1429~1.301\bin\Hostx64\x64\cl.exe  /nologo /TP -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_CHRONO_NO_LIB -DBOOST_CONTAINER_DYN_LINK -DBOOST_CONTAINER_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK=1 -DBOOST_FILESYSTEM_NO_LIB -DBOOST_GRAPH_DYN_LINK -DBOOST_GRAPH_NO_LIB -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_IOSTREAMS_NO_LIB -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_NO_LIB -DBOOST_RANDOM_DYN_LINK -DBOOST_RANDOM_NO_LIB -DBOOST_SERIALIZATION_DYN_LINK -DBOOST_SERIALIZATION_NO_LIB -DBOOST_STACKTRACE_DYN_LINK -DBOOST_STACKTRACE_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DBOOST_THREAD_USE_DLL -DBT_USE_DOUBLE_PRECISION -DBoost_USE_MULTITHREADED=ON -DBoost_USE_STATIC_LIBS=OFF -DBoost_USE_STATIC_RUNTIME=OFF -DEIGEN_HAS_CXX17_OVERALIGN=0 -DKISSFFT_DLL_IMPORT=1 -DTESSERACT_ASSIMP_USE_PBRMATERIAL=1 -DTESSERACT_PARSE_POINT_CLOUDS=\"ON\" -DTESSERACT_TASK_COMPOSER_DIR=\"C:/dev/tract/install/share/tesseract_task_composer\" -DTINYXML2_IMPORT -DTRAJOPT_DATA_DIR=\"C:/dev/tract/install/share/trajopt_common/data\" -D_FILE_OFFSET_BITS=64 -D_USE_MATH_DEFINES -D_USE_MATH_DEFINES=ON -D__SSE2__ -D__SSE3__ -D__SSE4_1__ -D__SSE4_2__ -D__SSE__ -D__SSSE3__ -Dkiss_fft_scalar=double -Dtesseract_task_composer_planning_factories_EXPORTS -IC:\dev\tract\tesseract_planning\tesseract_task_composer\planning\include -IC:\dev\tract\tesseract_planning\tesseract_task_composer\core\include -IC:\dev\tract\build\tesseract_task_composer\.. -external:I"C:\Program Files\PCL 1.14.0\3rdParty\Eigen3\include\eigen3" -external:IE:\programs\vcpkg\installed\x64-windows\include -external:IC:\dev\tract\install\include -external:IE:\programs\vcpkg\installed\x64-windows\include\eigen3 -external:IE:\programs\vcpkg\installed\x64-windows\include\bullet -external:I"C:\Program Files\OpenNI2\Include" -external:I"C:\Program Files\PCL 1.14.0\include\pcl-1.14" -external:I"C:\Program Files\PCL 1.14.0\3rdParty\VTK\include\vtk-9.3" -external:I"C:\Program Files\PCL 1.14.0\3rdParty\VTK\include\vtk-9.3\vtknlohmannjson\include" -external:I"C:\Program Files\PCL 1.14.0\3rdParty\VTK\include\vtk-9.3\vtkfreetype\include" -external:W0 /DWIN32 /D_WINDOWS /GR /EHsc /O2 /Ob2 /DNDEBUG -std:c++17 -MD -openmp /arch:AVX2 /showIncludes /Foplanning\CMakeFiles\tesseract_task_composer_planning_factories.dir\src\factories\planning_task_composer_ruckig_plugin_factory.cpp.obj /Fdplanning\CMakeFiles\tesseract_task_composer_planning_factories.dir\ /FS -c C:\dev\tract\tesseract_planning\tesseract_task_composer\planning\src\factories\planning_task_composer_ruckig_plugin_factory.cpp
C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.29.30133\include\memory(3089): error C2027: 使用了未定义类型“tesseract_planning::TaskComposerNodeInfo”
C:\dev\tract\tesseract_planning\tesseract_task_composer\core\include\tesseract_task_composer/core/task_composer_node.h(56): note: 参见“tesseract_planning::TaskComposerNodeInfo”的 声明
C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.29.30133\include\memory(3088): note: 在编译 类 模板 成员函数“void std::default_delete<tesseract_planning::TaskComposerNodeInfo>::operator ()(_Ty *) noexcept const”时
        with
        [
            _Ty=tesseract_planning::TaskComposerNodeInfo
        ]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.29.30133\include\memory(3198): note: 查看对正在编译的函数 模板 实例化“void std::default_delete<tesseract_planning::TaskComposerNodeInfo>::operator ()(_Ty *) noexcept const”的引用
        with
        [
            _Ty=tesseract_planning::TaskComposerNodeInfo
        ]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.29.30133\include\memory(3125): note: 查看对正在编译的 类 模板 实例化“std::default_delete<tesseract_planning::TaskComposerNodeInfo>”的引用
C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.29.30133\include\utility(325): note: 查看对正在编译的 类 模板 实例化“std::unique_ptr<tesseract_planning::TaskComposerNodeInfo,std::default_delete<tesseract_planning::TaskComposerNodeInfo>>”的引用
C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.29.30133\include\xtree(331): note: 查看对正在编译的 类 模板 实例化“std::pair<const boost::uuids::uuid,std::unique_ptr<tesseract_planning::TaskComposerNodeInfo,std::default_delete<tesseract_planning::TaskComposerNodeInfo>>>”的引用
C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.29.30133\include\xtree(752): note: 查看对正在编译的 类 模板 实例化“std::_Tree_node<std::pair<const boost::uuids::uuid,std::unique_ptr<tesseract_planning::TaskComposerNodeInfo,std::default_delete<tesseract_planning::TaskComposerNodeInfo>>>,std::_Default_allocator_traits<_Alloc>::void_pointer>”的引用

xbkaishui, 2024-08-26:
hello， any update on this

Levi-Armstrong, 2024-08-26:
Unfortunately, I am unable to assist with windows related as I only develop on linux. @johnwason are you able to help?

johnwason, 2024-08-26:
I can't read the error messages because they are not in English but I suspect there is a version mismatch with tesseract_python and the other packages. Because the tesseract API is unstable tesseract_python needs to use specific versions.

xbkaishui, 2024-08-27:
hello:

thanks for your quickly response. there is an issue related to tesseract_python package.  one question is can I use latest tesseract repo on tesseract_python?

xbkaishui, 2024-08-28:
ok, I can submit a pr, can you give a summary about which api is changed ?

xbkaishui, 2024-08-28:
I have readed tesseract code,  this is a great project. thanks

johnwason, 2024-08-28:
What issue are you having? It may be easier to cherrypick the fix than update the python code.

xbkaishui, 2024-08-29:
the kdl IK solver success rate is too low for 6 axis arm. I am using aubo i5,  I want to upgrade the ik sovler to kdl-RR (random restart), implement by  track_ik.

johnwason, 2024-08-29:
That looks like a UR clone. There is an analytical UR solver available in the Python package.

johnwason, 2024-08-30:
@Levi-Armstrong can you help set up the UR invkin plugin configuration? The aubo i5 looks like a UR clone. Not sure about the kinematic parameters.

johnwason, 2024-08-30:
@xbkaishu the solver you are trying to use is a numeric solver that iteratively tries to find a solution. For special robot designs like "spherical wrist"/OPW robots, and Universal Robots Tesseract has a closed-form analytical solution that can directly compute the kinematics without needing to use an iterative solution.

xbkaishui, 2024-08-30:
Got it @johnwason  Thanks,  @Levi-Armstrong please help

xbkaishui, 2024-08-30:
> @xbkaishu the solver you are trying to use is a numeric solver that iteratively tries to find a solution. For special robot designs like "spherical wrist"/OPW robots, and Universal Robots Tesseract [...]

by the way, can i submit a pr for new implement kdl-RR (random restart), I test it the success rate is higher than kdl-nr and kdl-lma

Levi-Armstrong, 2025-01-04:
> by the way, can i submit a pr for new implement kdl-RR (random restart), I test it the success rate is higher than kdl-nr and kdl-lma

Absolutely you can submit a PR adding these solvers.

closed 2025-01-15 by xbkaishui