tesseract-robotics/tesseract_planning #510: Can I add constraints in trajopt?

sm3304love, 2024-08-27:
Hello,

I want to use TrajOpt in my project and have some questions.

I would like to know how I can add constraints, such as joint velocity and Cartesian velocity, to my plan profile.

I noticed the `createJointVelocityConstraint` function in `trajopt_ifopt_utils.h`, but I couldn't find any examples demonstrating how to set a joint velocity constraint using `createJointVelocityConstraint`.

Additionally, I couldn't find any function in `trajopt_ifopt_utils.h` to set Cartesian velocity constraints. Is there no option to set Cartesian velocity constraints in TrajOpt?

Thank you.

SiruiXie, 2024-12-17:
Hi, I ran into that very same issue recently. Have you found any possible solution? @sm3304love

rjoomen, 2024-12-17:
I'm only aware of [the tcpSpeedLimiter](https://github.com/ros-industrial-consortium/scan_n_plan_workshop/issues/50) in Scan 'n Plan. Somewhat related is the [Constant TCP velocity time parameterization](https://github.com/ros-industrial-consortium/scan_n_plan_workshop/pull/28).