tesseract-robotics/tesseract_planning #531: Compile error when including tesseract_task_composer_planning_nodes_export.h

173hyj, 2024-11-19:
hello:
i am building planning master branch on ubuntu 20.04.6(focal)

command:
```bash
rosinstall src/tesseract_ros/ src/tesseract_ros/dependencies.rosinstall
rosdep install --from-paths src -iry
catkin_make_isolated
```
and when i run `catkin_make_isolated`, there print some error
```log
==> make -j16 -l16 in '/home/hyj/colcon_ws/build_isolated/tesseract_task_composer/devel'
[ 23%] Built target tesseract_task_composer
[ 24%] Building CXX object core/CMakeFiles/tesseract_task_composer_nodes.dir/src/nodes/has_data_storage_entry_task.cpp.o
[ 29%] Built target tesseract_task_composer_taskflow
[ 32%] Built target tesseract_task_composer_taskflow_factories
[ 35%] Built target tesseract_task_composer_trajopt_example
[ 38%] Built target tesseract_task_composer_raster_example
[ 41%] Built target tesseract_task_composer_example
[ 43%] Linking CXX shared library libtesseract_task_composer_nodes.so
[ 52%] Built target tesseract_task_composer_nodes
Scanning dependencies of target tesseract_task_composer_factories
Scanning dependencies of target tesseract_task_composer_planning_nodes
[ 53%] Building CXX object core/CMakeFiles/tesseract_task_composer_factories.dir/src/task_composer_task_plugin_factory.cpp.o
[ 55%] Building CXX object planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/src/nodes/continuous_contact_check_task.cpp.o
[ 56%] Building CXX object planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/src/nodes/fix_state_bounds_task.cpp.o
[ 58%] Building CXX object planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/src/nodes/format_as_input_task.cpp.o
[ 60%] Building CXX object planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/src/nodes/fix_state_collision_task.cpp.o
[ 61%] Building CXX object planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/src/nodes/format_as_result_task.cpp.o
[ 63%] Building CXX object planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/src/nodes/discrete_contact_check_task.cpp.o
[ 64%] Building CXX object planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/src/nodes/profile_switch_task.cpp.o
[ 66%] Building CXX object planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/src/nodes/min_length_task.cpp.o
[ 72%] Building CXX object planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/src/nodes/upsample_trajectory_task.cpp.o
[ 72%] Building CXX object planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/src/nodes/process_planning_input_task.cpp.o
[ 72%] Building CXX object planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/src/nodes/update_end_state_task.cpp.o
[ 72%] Building CXX object planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/src/nodes/update_start_and_end_state_task.cpp.o
[ 73%] Building CXX object planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/src/nodes/update_start_state_task.cpp.o
[ 76%] Building CXX object planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/src/nodes/raster_only_motion_task.cpp.o
[ 76%] Building CXX object planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/src/nodes/raster_motion_task.cpp.o
In file included from /home/hyj/colcon_ws/src/tesseract_ros/tesseract_planning/tesseract_task_composer/planning/src/nodes/update_start_and_end_state_task.cpp:34:
/home/hyj/colcon_ws/src/tesseract_ros/tesseract_planning/tesseract_task_composer/planning/include/tesseract_task_composer/planning/nodes/update_start_and_end_state_task.h:32:10: fatal error: tesseract_task_composer/planning/tesseract_task_composer_planning_nodes_export.h: No such file or directory
   32 | #include <tesseract_task_composer/planning/tesseract_task_composer_planning_nodes_export.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/build.make:193: planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/src/nodes/update_start_and_end_state_task.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/hyj/colcon_ws/src/tesseract_ros/tesseract_planning/tesseract_task_composer/planning/src/nodes/upsample_trajectory_task.cpp:34:
/home/hyj/colcon_ws/src/tesseract_ros/tesseract_planning/tesseract_task_composer/planning/include/tesseract_task_composer/planning/nodes/upsample_trajectory_task.h:32:10: fatal error: tesseract_task_composer/planning/tesseract_task_composer_planning_nodes_export.h: No such file or directory
   32 | #include <tesseract_task_composer/planning/tesseract_task_composer_planning_nodes_export.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/hyj/colcon_ws/src/tesseract_ros/tesseract_planning/tesseract_task_composer/planning/src/nodes/update_end_state_task.cpp:34:
/home/hyj/colcon_ws/src/tesseract_ros/tesseract_planning/tesseract_task_composer/planning/include/tesseract_task_composer/planning/nodes/update_end_state_task.h:32:10: fatal error: tesseract_task_composer/planning/tesseract_task_composer_planning_nodes_export.h: No such file or directory
   32 | #include <tesseract_task_composer/planning/tesseract_task_composer_planning_nodes_export.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/hyj/colcon_ws/src/tesseract_ros/tesseract_planning/tesseract_task_composer/planning/src/nodes/raster_motion_task.cpp:35:
/home/hyj/colcon_ws/src/tesseract_ros/tesseract_planning/tesseract_task_composer/planning/include/tesseract_task_composer/planning/nodes/raster_motion_task.h:34:10: fatal error: tesseract_task_composer/planning/tesseract_task_composer_planning_nodes_export.h: No such file or directory
   34 | #include <tesseract_task_composer/planning/tesseract_task_composer_planning_nodes_export.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/hyj/colcon_ws/src/tesseract_ros/tesseract_planning/tesseract_task_composer/planning/src/nodes/update_start_state_task.cpp:34:
/home/hyj/colcon_ws/src/tesseract_ros/tesseract_planning/tesseract_task_composer/planning/include/tesseract_task_composer/planning/nodes/update_start_state_task.h:32:10: fatal error: tesseract_task_composer/planning/tesseract_task_composer_planning_nodes_export.h: No such file or directory
   32 | #include <tesseract_task_composer/planning/tesseract_task_composer_planning_nodes_export.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/build.make:219: planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/src/nodes/upsample_trajectory_task.cpp.o] Error 1
make[2]: *** [planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/build.make:180: planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/src/nodes/update_end_state_task.cpp.o] Error 1
make[2]: *** [planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/build.make:232: planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/src/nodes/raster_motion_task.cpp.o] Error 1
make[2]: *** [planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/build.make:206: planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/src/nodes/update_start_state_task.cpp.o] Error 1
[... 43 lines cut ...]
/home/hyj/colcon_ws/src/tesseract_ros/tesseract_planning/tesseract_task_composer/planning/include/tesseract_task_composer/planning/nodes/discrete_contact_check_task.h:33:10: fatal error: tesseract_task_composer/planning/tesseract_task_composer_planning_nodes_export.h: No such file or directory
   33 | #include <tesseract_task_composer/planning/tesseract_task_composer_planning_nodes_export.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/build.make:76: planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/src/nodes/discrete_contact_check_task.cpp.o] Error 1
In file included from /home/hyj/colcon_ws/src/tesseract_ros/tesseract_planning/tesseract_task_composer/planning/src/nodes/continuous_contact_check_task.cpp:45:
[... cut ...]
make[1]: *** [CMakeFiles/Makefile2:281: planning/CMakeFiles/tesseract_task_composer_planning_nodes.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 78%] Linking CXX shared library libtesseract_task_composer_factories.so
[ 78%] Built target tesseract_task_composer_factories
make: *** [Makefile:130: all] Error 2
<== Failed to process package 'tesseract_task_composer':
  Command '['/home/hyj/colcon_ws/devel_isolated/tesseract_motion_planners/env.sh', 'make', '-j16', '-l16']' returned non-zero exit status 2.
```

CodingHzx, 2024-11-20:
oh, It seems like you're facing the same issue. 
You can use `find -name "tesseract_task_composer_nodes_export.h" `to check if the header file has been generated.

173hyj, 2024-11-20:
> oh, It seems like you're facing the same issue. You can use `find -name "tesseract_task_composer_nodes_export.h" `to check if the header file has been generated.

there found a file with that name.
```bash
$ sudo find . -name "tesseract_task_composer_nodes_export.h"
./build_isolated/tesseract_task_composer/devel/core/tesseract_task_composer_nodes_export.h
```