tesseract-robotics/tesseract_planning #593: Is it necessary to integrate a function similar to ‘ros control’ to communicate and control the actual robot?

liuminL, 2025-02-21:
The result of the planning is some waypoints, which are not dense. Is it necessary to use some code for interpolation here? If the interpolated trajectory is to communicate and control the actual robot, issues such as real-time performance and interface matching need to be considered. Is it necessary to construct an integrated controller architecture similar to [**"ros controls"**](https://github.com/ros-controls/ros2_control)?

schornakj, 2025-03-17:
> The result of the planning is some waypoints, which are not dense. Is it necessary to use some code for interpolation here?

It depends on how your hardware handles trajectories, but usually it's desirable to upsample the trajectory. One way to do this is to add [an UpsampleTrajectoryTask node](https://github.com/tesseract-robotics/tesseract_planning/blob/master/tesseract_task_composer/planning/include/tesseract_task_composer/planning/nodes/upsample_trajectory_task.h) to your task config. This adds new waypoints to your trajectory according to a minimum segment length setting.

> Is it necessary to construct an integrated controller architecture similar to ["ros controls"](https://github.com/ros-controls/ros2_control)?

Tesseract's point of interface for trajectory execution is usually a control_msgs/FollowJointTrajectory action advertised by a server which talks to the robot hardware. Whether ros2_control specifically is the right answer depends on your use case and hardware, but speaking generally you will need something specific to your robot that reads the trajectory message and generates control inputs to the robot's joints that make it follow the trajectory.

schornakj, 2025-03-19:
[Kinova does have a ROS 2 driver package](https://github.com/Kinovarobotics/ros2_kortex) that communicates with the robot through its lower-level API, so if you're using one of their robots you may want to explore over there.

liuminL, 2025-04-01:
thank you very much