tesseract-robotics/tesseract_planning #601: Avoid using `assert` for validation during runtime

schornakj, 2025-03-17:
In general I think it's a good idea to avoid using `assert` for things like enforcing that a joint trajectory satisfies the kinematic chain's joint limit.

[Here's an example in the Descartes planner](https://github.com/tesseract-robotics/tesseract_planning/blob/189269eb9b38b22be7b52ea1bd422f432dec0a29/tesseract_motion_planners/descartes/include/tesseract_motion_planners/descartes/impl/descartes_motion_planner.hpp#L160) -- it's possible to hit this assert even if other plans have succeeded during the same session if the program is composing joint waypoints at runtime.

Instead, these cases should create an error message describing what happened and return a failure status. This gives an opportunity for higher-level control logic to respond to the failure automatically if appropriate.

Happy to  write up some PRs for this myself, just wanted to post an issue about it before I got started.

Levi-Armstrong, 2025-03-17:
I think we made this change in some of the other planners which had this check.

Levi-Armstrong, 2025-04-20:
Addressed by #606

closed 2025-04-20 by Levi-Armstrong