tesseract-robotics/tesseract_planning #602: If the specified point will collide, are there parameters that allow the attitude not to reach the specified point in the middle

Jaguar1995, 2025-03-18:
I have a need to specify a series of points, the position of the points required to achieve, by changing the attitude to avoid collision. Are there any relevant examples, please

Levi-Armstrong, 2025-03-18:
This done in the [puzzle piece example](https://github.com/tesseract-robotics/tesseract_planning/blob/189269eb9b38b22be7b52ea1bd422f432dec0a29/tesseract_examples/src/puzzle_piece_example.cpp#L225-L231) where we setup the trajopt profile to allow it to rotate around the z-axis.

Jaguar1995, 2025-03-19:
Thank you very much, can I individually set one or the last pose in a series of poses that must be reached with cartesian_constraint_config.coeff(5)=0

Jaguar1995, 2025-03-20:
Thank you for your answer, there are two questions:
(1) According to what you said, I set different "PlanProfile" for the starting point and the end point respectively, and added different "plan_profile" to "profiles", but the planning failed, and the final trajectory was collision. I modified the program in glass_upright_example.cpp, I uploaded it, please check if there is any problem.
(2) Is the attitude between the specified two poses similar to linear interpolation, can only slowly change from one attitude to the other, and can not, for example, first turn 30 degrees positive about the y axis from the starting attitude, and then reverse 30 degrees about the y axis? @Levi-Armstrong

Jaguar1995, 2025-03-20:
`/**
 * @file glass_upright_example.cpp
 * @brief Glass upright example implementation
 *
 * @author Levi Armstrong
 * @date July 22, 2019
 * @version TODO
 * @bug No known bugs
 *
 * @copyright Copyright (c) 2017, Southwest Research Institute
 *
 * @par License
 * Software License Agreement (Apache License)
 * @par
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 * http://www.apache.org/licenses/LICENSE-2.0
 * @par
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */
#include <tesseract_common/macros.h>
TESSERACT_COMMON_IGNORE_WARNINGS_PUSH
#include <json/json.h>
#include <console_bridge/console.h>
#include <trajopt_common/collision_types.h>
#include <trajopt/problem_description.hpp>
TESSERACT_COMMON_IGNORE_WARNINGS_POP

#include <tesseract_examples/glass_upright_example.h>

#include <tesseract_common/resource_locator.h>

#include <tesseract_scene_graph/link.h>
#include <tesseract_scene_graph/joint.h>

#include <tesseract_state_solver/state_solver.h>

#include <tesseract_environment/environment.h>
#include <tesseract_environment/commands/add_link_command.h>
#include <tesseract_environment/utils.h>

#include <tesseract_common/stopwatch.h>

#include <tesseract_command_language/profile_dictionary.h>
#include <tesseract_command_language/composite_instruction.h>
#include <tesseract_command_language/state_waypoint.h>
#include <tesseract_command_language/cartesian_waypoint.h>
#include <tesseract_command_language/joint_waypoint.h>
#include <tesseract_command_language/move_instruction.h>
#include <tesseract_command_language/utils.h>

#include <tesseract_task_composer/core/task_composer_context.h>
#include <tesseract_task_composer/core/task_composer_data_storage.h>
#include <tesseract_task_composer/core/task_composer_node.h>
#include <tesseract_task_composer/core/task_composer_executor.h>
#include <tesseract_task_composer/core/task_composer_future.h>
#include <tesseract_task_composer/core/task_composer_plugin_factory.h>

#include <tesseract_visualization/visualization.h>
#include <tesseract_visualization/markers/toolpath_marker.h>

#include <tesseract_motion_planners/core/utils.h>
#include <tesseract_motion_planners/trajopt_ifopt/profile/trajopt_ifopt_default_composite_profile.h>
#include <tesseract_motion_planners/trajopt/profile/trajopt_default_composite_profile.h>
#include <tesseract_motion_planners/trajopt_ifopt/profile/trajopt_ifopt_default_plan_profile.h>
[... 253 lines cut ...]
    auto ci = future->context->data_storage->getData(output_key).as<CompositeInstruction>();
    tesseract_common::Toolpath toolpath = toToolpath(ci, *env_);
    // print toolpath
    for (const auto& wp : toolpath)
    {
      std::cout << "Translation of Waypoint: " << std::endl;
      for (int i = 0; i < wp.size(); ++i) {
        std::cout << "[" << wp[i].translation()[0] << ", "<< wp[i].translation()[1]<< ", " << wp[i].translation()[2]<< ", ";
        Eigen::Matrix3d rotation_matrix = wp[i].rotation();
        Eigen::Quaterniond q(rotation_matrix);
        std::cout << q.x() << ", " << q.y() << ", " << q.z() << ", "<< q.w() << "]," << std::endl;        
      }
    }
  }
  if (future->context->isSuccessful()) {
    CONSOLE_BRIDGE_logInform("Final trajectory is collision free");
  } else {
    CONSOLE_BRIDGE_logError("Final trajectory is not collision free");
  }
  // Check if final trajectory is collision free
  // CONSOLE_BRIDGE_logInform("Final trajectory is collision free");
  return future->context->isSuccessful();
}
}  // namespace tesseract_examples
`

Levi-Armstrong, 2025-04-20:
The Move Instruction has two types of profiles. The Profile is specific to the waypoint its self and the Path Profile is specific to the interpolated point from the previous waypoint to this waypoint. When you construct the waypoint given a single profile it assigns this provide to both the Profile and Path Profile. Since the target profile is fully constrained then all interpolated points between the START and TARGET will use the TARGET profile. If you still want the glass upright behavior you need to set the path profile to UPRIGHT for the target move instruction.