tesseract-robotics/tesseract_planning #609: Crash on destruction of TrajOptIfoptOSQPSolverProfile

rjoomen, 2025-04-16:
### Version

latest

### OS Version

Ubuntu 24.04

### Describe the bug

The default move constructor and move assign operator of TrajOptIfoptOSQPSolverProfile cause a "double free or corruption (!prev)" crash on destruction of the class if it was moved, for instance through serialization.

I believe [this PR](https://github.com/tesseract-robotics/tesseract_planning/pull/603) is the culprit. Deleting the copy constructors is correct, but I believe qp_settings should stay a unique_ptr, which can be moved but not copied.

I tested this and it solves the issue.

### To Reproduce

Move TrajOptIfoptOSQPSolverProfile.

### Expected behavior

No crash.

### Relevant log output

```shell

```

Levi-Armstrong, 2025-04-16:
Is qp_settings not movable?

rjoomen, 2025-04-17:
Seems like it. And osqp-eigen itself is also using the settings [using a unique_ptr](https://github.com/robotology/osqp-eigen/blob/master/include/OsqpEigen/Solver.hpp#L44), so why shouldn't we?

PR #610 (2025-04-19): Store OSQPEigenSettings as unique_ptr
Partially reverts #603. Fixes #609
closed 2025-04-19 by Levi-Armstrong