tesseract-robotics/tesseract_planning #615: Make loading of default profiles optional

marip8, 2025-04-23:
### Is your feature request related to a problem? Please describe.

As of 0.29.x, when a task cannot find a specified profile, it chooses a default profile instead of returning an error. This can result in undesired planning behavior that can be hard to track down.

### Describe the solution you'd like

1. Remove [the `getProfile` helper function](https://github.com/tesseract-robotics/tesseract_planning/blob/671ea7897623a90179314a624835dea67c66c388/tesseract_motion_planners/core/include/tesseract_motion_planners/planner_utils.h#L109) that returns a provided default profile if the desired profile is not available
2. Add a helper function that adds a default profiles for each task to a profile dictionary. Then the user can choose whether or not to populate their profile dictionary with default profiles or if they want to be responsible for adding them all themselves
    - One approach (implemented [here](https://github.com/marip8/tesseract_planning/commit/d5d7acc6069e2b6363b7ec0652c642e0654086aa)) is to write a where we manually a default profile for all known tasks. The limitation here is that this would only add default profiles for tasks that exist in Tesseract, and not user provided tasks.
    - Probably a better approach (discussed [here](https://github.com/tesseract-robotics/tesseract_planning/pull/367#issuecomment-1724561496)) is to add a function to the task composer task base class that generates a default profile for that task. Then our helper function could loop over all known task composer plugins and invoke the task method for generating the default profile. Then we would be able to add default profiles for all findable (and therefore usable) tasks.

### Describe alternatives you've considered

N/A

Levi-Armstrong, 2025-04-24:
@marip8 Do you think this will still be needed once we move to leveraging a yaml file to load the profiles as plugins?

marip8, 2025-04-24:
I think so. Loading profiles via YAML won't really change how the profiles are found by the tasks. That will make the profiles easier to specify them at least, but it would still be a convenient feature for new Tesseract users to ensure things can work without having to write an extensive YAML file