tesseract-robotics/tesseract_planning #621: TrajOptIfoptDefaultMoveProfile resets Cartesian cost coeffs to 1

rjoomen, 2025-05-09:
See [here](https://github.com/tesseract-robotics/tesseract_planning/blob/master/tesseract_motion_planners/trajopt_ifopt/src/profile/trajopt_ifopt_default_move_profile.cpp#L121). Why should Cartesian coefficients be reset to 1? Seems wrong to me.

Levi-Armstrong, 2025-05-12:
I do agree this needs some further investigation. Bellow is what was used before, but not sure which one is correct. There are difference with how/when the coeff is applied which needs to be revised to make sure this is correct and documented. 

[addCartesianPositionSquaredCost](https://github.com/tesseract-robotics/tesseract_planning/blob/0.27.0/tesseract_motion_planners/trajopt_ifopt/src/trajopt_ifopt_utils.cpp)

Levi-Armstrong, 2025-05-12:
We really need a good way to compare the two versions.

Levi-Armstrong, 2025-05-12:
If you search in trajopt_qp_problem.cpp for weight, you will find a few todo that likely need to be addressed. I think the easiest way to figure a good way to compare the osqp matrices between Legacy TrajOpt and Ifopt to verify they align.

Levi-Armstrong, 2025-05-12:
The legacy trajopt handles the cartesian cost using a cost from error function [here](https://github.com/tesseract-robotics/trajopt/blob/dd609dd688d4906566d72814e6c282f2b094b97d/trajopt_sco/src/modeling_utils.cpp#L168).