tesseract-robotics/tesseract_planning #631: Task Composer Plugin Factory increasing constructor time on repeat calls

marrts, 2025-06-25:
### Version

master

### OS Version

Ubuntu 22.04

### Describe the bug

When constructing a `TaskComposerPluginFactory` with a re-used `YAML::Node`, each successive call to the constructor takes longer than the previous call. 

### To Reproduce

I added some tests to `tesseract_task_composer_plugin_factories_unit.cpp` on this branch on my fork: https://github.com/marrts/tesseract_planning/tree/bug/task_composer_plugin_factory_increasing_time

Look below for the output of the tests:

### Expected behavior

Each time calling the constructor should be consistent in the time it takes to construct the object. I think there is some weird YAML shared memory stuff going on.

### Relevant log output

```shell
[ RUN      ] TesseractTaskComposerFactoryUnit.LoadTaskComposerPluginFactoryMultiTimes
LoadFromYAMLNode - Iteration 0: 0.638938 seconds
LoadFromYAMLNode - Iteration 1: 1.01681 seconds
LoadFromYAMLNode - Iteration 2: 1.53459 seconds
LoadFromYAMLNode - Iteration 3: 2.00653 seconds
LoadFromYAMLNode - Iteration 4: 2.40333 seconds
LoadFromYAMLNode - Iteration 5: 2.90787 seconds
LoadFromYAMLNode - Iteration 6: 3.17379 seconds
LoadFromYAMLNode - Iteration 7: 3.65178 seconds
LoadFromYAMLNode - Iteration 8: 4.09297 seconds
LoadFromYAMLNode - Iteration 9: 4.53538 seconds
/tesseract_planning/tesseract_task_composer/test/tesseract_task_composer_plugin_factories_unit.cpp:334: Failure
Expected: (slope) < (mean * 0.10), actual: 0.431475 vs 0.25962
Detected a significant upward trend in timings (slope=0.43147530670302991) for test: LoadFromYAMLNode
[  FAILED  ] TesseractTaskComposerFactoryUnit.LoadTaskComposerPluginFactoryMultiTimes (25978 ms)
[ RUN      ] TesseractTaskComposerFactoryUnit.LoadTaskComposerPluginFactoryMultiTimesInLambda
LoadFromYAMLNodeInLambda - Iteration 0: 0.642663 seconds
LoadFromYAMLNodeInLambda - Iteration 1: 0.642031 seconds
LoadFromYAMLNodeInLambda - Iteration 2: 0.634964 seconds
LoadFromYAMLNodeInLambda - Iteration 3: 0.63679 seconds
LoadFromYAMLNodeInLambda - Iteration 4: 0.643559 seconds
LoadFromYAMLNodeInLambda - Iteration 5: 0.63511 seconds
LoadFromYAMLNodeInLambda - Iteration 6: 0.645802 seconds
LoadFromYAMLNodeInLambda - Iteration 7: 0.638519 seconds
LoadFromYAMLNodeInLambda - Iteration 8: 0.642491 seconds
LoadFromYAMLNodeInLambda - Iteration 9: 0.637839 seconds
[       OK ] TesseractTaskComposerFactoryUnit.LoadTaskComposerPluginFactoryMultiTimesInLambda (6411 ms)
[ RUN      ] TesseractTaskComposerFactoryUnit.LoadTaskComposerPluginFactoryMultiTimesFromFile
LoadFromFile - Iteration 0: 0.638614 seconds
LoadFromFile - Iteration 1: 0.643443 seconds
LoadFromFile - Iteration 2: 0.631421 seconds
LoadFromFile - Iteration 3: 0.639436 seconds
LoadFromFile - Iteration 4: 0.632986 seconds
LoadFromFile - Iteration 5: 0.637477 seconds
LoadFromFile - Iteration 6: 0.639279 seconds
LoadFromFile - Iteration 7: 0.62811 seconds
LoadFromFile - Iteration 8: 0.637722 seconds
LoadFromFile - Iteration 9: 0.639037 seconds
[       OK ] TesseractTaskComposerFactoryUnit.LoadTaskComposerPluginFactoryMultiTimesFromFile (6373 ms)
[ RUN      ] TesseractTaskComposerFactoryUnit.LoadTaskComposerPluginFactoryMultiTimesFromString
LoadFromString - Iteration 0: 0.640081 seconds
LoadFromString - Iteration 1: 0.645035 seconds
LoadFromString - Iteration 2: 0.639489 seconds
LoadFromString - Iteration 3: 0.631591 seconds
LoadFromString - Iteration 4: 0.643855 seconds
LoadFromString - Iteration 5: 0.628335 seconds
LoadFromString - Iteration 6: 0.646938 seconds
LoadFromString - Iteration 7: 0.645157 seconds
LoadFromString - Iteration 8: 0.639327 seconds
LoadFromString - Iteration 9: 0.639432 seconds
[       OK ] TesseractTaskComposerFactoryUnit.LoadTaskComposerPluginFactoryMultiTimesFromString (6406 ms)
[ RUN      ] TesseractTaskComposerFactoryUnit.LoadTaskComposerPluginFactoryMultiTimesFromFileWithLoadYamlFile
LoadFromFileWithLoadYamlFile - Iteration 0: 1.71758 seconds
LoadFromFileWithLoadYamlFile - Iteration 1: 1.69915 seconds
LoadFromFileWithLoadYamlFile - Iteration 2: 1.70153 seconds
LoadFromFileWithLoadYamlFile - Iteration 3: 1.70419 seconds
LoadFromFileWithLoadYamlFile - Iteration 4: 1.70113 seconds
LoadFromFileWithLoadYamlFile - Iteration 5: 1.71002 seconds
LoadFromFileWithLoadYamlFile - Iteration 6: 1.71234 seconds
LoadFromFileWithLoadYamlFile - Iteration 7: 1.70967 seconds
LoadFromFileWithLoadYamlFile - Iteration 8: 1.70138 seconds
LoadFromFileWithLoadYamlFile - Iteration 9: 1.69878 seconds
[       OK ] TesseractTaskComposerFactoryUnit.LoadTaskComposerPluginFactoryMultiTimesFromFileWithLoadYamlFile (17063 ms)
[ RUN      ] TesseractTaskComposerFactoryUnit.LoadTaskComposerPluginFactoryMultiTimesFromFileWithLoadYamlFileOutside
LoadFromFileWithLoadYamlFileOutside - Iteration 0: 1.07564 seconds
LoadFromFileWithLoadYamlFileOutside - Iteration 1: 1.58299 seconds
LoadFromFileWithLoadYamlFileOutside - Iteration 2: 2.12883 seconds
LoadFromFileWithLoadYamlFileOutside - Iteration 3: 2.60944 seconds
LoadFromFileWithLoadYamlFileOutside - Iteration 4: 3.04392 seconds
LoadFromFileWithLoadYamlFileOutside - Iteration 5: 3.57309 seconds
LoadFromFileWithLoadYamlFileOutside - Iteration 6: 3.93509 seconds
LoadFromFileWithLoadYamlFileOutside - Iteration 7: 4.28796 seconds
LoadFromFileWithLoadYamlFileOutside - Iteration 8: 4.78774 seconds
LoadFromFileWithLoadYamlFileOutside - Iteration 9: 5.26942 seconds
/tesseract_planning/tesseract_task_composer/test/tesseract_task_composer_plugin_factories_unit.cpp:334: Failure
Expected: (slope) < (mean * 0.10), actual: 0.457449 vs 0.322941
Detected a significant upward trend in timings (slope=0.45744854383030265) for test: LoadFromFileWithLoadYamlFileOutside
[  FAILED  ] TesseractTaskComposerFactoryUnit.LoadTaskComposerPluginFactoryMultiTimesFromFileWithLoadYamlFileOutside (32946 ms)
```

Levi-Armstrong, 2025-06-25:
The one thing that constructor calls that the others do not is likely causing the issue `processYamlIncludeDirective` call.

Levi-Armstrong, 2025-06-25:
@marrts Can you test with [this](https://github.com/tesseract-robotics/tesseract/pull/1166) change to tesseract?

marrts, 2025-06-25:
> @marrts Can you test with https://github.com/tesseract-robotics/tesseract/pull/1166 change to tesseract?

That does not seem to fix the problem

marrts, 2025-06-25:
That idea got me thinking though and I can confirm it is fixed with:

```c++
void TaskComposerPluginFactory::loadConfig(const YAML::Node& config, const tesseract_common::ResourceLocator& locator)
{
  loadConfig(tesseract_common::processYamlIncludeDirective(YAML::Clone(config), locator));
}
```

Levi-Armstrong, 2025-07-05:
Addressed by this [PR](https://github.com/tesseract-robotics/tesseract/pull/1172)

closed 2025-07-05 by Levi-Armstrong