tesseract-robotics/tesseract_planning #665: Trajopt enforces joint limits after optimization

rjoomen, 2025-09-01:
Trajopt bluntly enforces joint limits after the optimization [here](https://github.com/tesseract-robotics/tesseract_planning/blob/master/tesseract_motion_planners/trajopt/src/trajopt_motion_planner.cpp#L158). Shouldn't the optimization itself take care of this? And in case this fails or if a user has some other reason to do a simple joint limit enforcement, shouldn't something like the [fix_state_bounds_profile.](https://github.com/tesseract-robotics/tesseract_planning/blob/master/tesseract_task_composer/planning/src/profiles/fix_state_bounds_profile.cpp) or the [kinematic_limits_check_profile](https://github.com/tesseract-robotics/tesseract_planning/blob/master/tesseract_task_composer/planning/src/profiles/kinematic_limits_check_profile.cpp) be used?

Levi-Armstrong, 2025-09-02:
This was added because trajopt was producing solution slightly outside of the limits causing issues down stream. I believe this should be handled by trajopt because I think this is a numerical issue and it just needs to be clamped to avoid issues down stream.

rjoomen, 2025-09-03:
A, I guess it's when it optimizes against a joint limit it can be over the limit by the constraint tolerance. Not a numerical issue then, but a feature. Would it be an idea to automatically subtract (or add, for a lower limit) the constraint tolerance from the constraints, so as to ensure the optimizer (and hence the robot) stays within the limits? Then this limit enforcement isn't necessary anymore either.

This also happens for collisions: a collision check after Trajopt has ran sometimes still finds small collisions, but less than the constraint tolerance within the collision margin.

Levi-Armstrong, 2025-09-04:
> But shouldn't this fixed joint be not part of the optimization in the first place, as it's fixed?

I believe in the legacy trajopt they are treated as soft constraints, but I believe in the ifopt version they are treated as a hard constraints.

> So having the limits corrected automatically by Trajopt would be a nice feature.

I agree, should the code for enforcing the limits just be moved to trajopt? I would recommend doing what is cleanest. 

``` c++
for (Eigen::Index i = 0; i < traj.rows(); i++)
  {
    assert(tesseract_common::satisfiesLimits<double>(traj.row(i), joint_limits, 1e-4));
    tesseract_common::enforceLimits<double>(traj.row(i), joint_limits);
  }
```